ros-drivers/transport_drivers #11: serial_driver only supports one type of message

JWhitleyWork, 2020-03-05:
Similar to #3 but for `serial_driver`. See https://github.com/ros-drivers/transport_drivers/blob/7660c2a02700ffafe8fc0839d45ff688c1a31db4/serial_driver/include/serial_driver/serial_driver_node.hpp#L64.

For some sensors, it'd be useful to be able to generate messages of different types

JWhitleyWork, 2020-03-05:
Previous discussion: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/issues/132

flynneva, 2021-01-28:
wouldnt it make sense for this package to purely handle grabbing data from the serial port and then publishing it to a ROS2 topic (uint8_t array or something) and then anyone could write a package that subscribes to that topic and does whatever they want with it?

what I mean by this question is isnt this the intended use of this `serial_driver` package?

JWhitleyWork, 2021-01-28:
@flynneva While this is the way that the ROS1 implementation operated, the advantage here is that you don't have to deal with the additional overhead and delay of publishing each serial message. This allows you to simply inherit from this class and get all of the I/O functionality that you're looking for built into your node in a cross-platform manner.

flynneva, 2021-01-28:
@JWhitleyWork so sort of like a psuedo pure cpp library for interfacing with serial devices....just specific for ROS2?

JWhitleyWork, 2021-01-28:
Sort of, yes. I'm assuming by "psuedo-pure-cpp" you mean "using Boost."

flynneva, 2021-01-28:
are there any examples of people using this library yet? ive been going off of what is in the `test` dir but would be cool to see it in action in the real world too.

JWhitleyWork, 2021-01-28:
Autoware.Auto currently uses the released version. You can see an example in https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/drivers/xsens_nodes/include/xsens_nodes/xsens_common_node.hpp and a UDP example in https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/drivers/velodyne_nodes/include/velodyne_nodes/velodyne_cloud_node.hpp.

flynneva, 2021-02-01:
so i guess my question is what is left to do if people can already use the library as desired?

if you want a generic serial package that works for all serial devices you'll have to just create a node using the library that publishes the data to a topic with a uint8_t array type or something....that would probably be the easiest way to do it. is that what people are asking for? that still only supports one type of message I guess....

flynneva, 2021-03-09:
@JWhitleyWork for the `udp_driver` we got lucky that I had already released the `udp_msgs` package. should I get to work to develop a `serial_msgs` package as well?

flynneva, 2021-03-09:
I [created a repo really quickly for `serial_msgs`](https://github.com/flynneva/serial_msgs) probably missing something but feel free to check it out. I can make a release pretty quickly once given the "green light".

JWhitleyWork, 2021-08-31:
This is no longer the case since the major code reorg for 1.0. Closing.

closed 2021-08-31 by JWhitleyWork