ros-drivers/transport_drivers #12: Add ability to send UDP/Serial data

JWhitleyWork, 2020-03-25:
The current implementations in `udp_driver` and `serial_driver` only support receiving data.

flynneva, 2021-01-27:
@JWhitleyWork awhile back you reached out to me about helping out with this project. ive found some time this week and id like to help out bring this up. im going to first focus on the `udp_driver` if thats ok....then ill help out wiht the `serial_driver` if I can get it done quickly.

JWhitleyWork, 2021-01-27:
@flynneva Thanks for working on this! Much appreciated. We actually are in desperate need of the ability to send serial data right now for the F1TENTH community (https://f1tenth.org/).

flynneva, 2021-01-27:
yeah I reached out to Kat on Twitter. Was planning on tackling that tomorrow actually (getting some bno055 boards in the mail tomorrow that use serial). I'll see what I can do.

JWhitleyWork, 2021-01-27:
Very cool, thanks. If you need me to test anything on some hardware here, I can do that too. I have an F1TENTH car and a few VESC ESCs that communicate over serial.

JWhitleyWork, 2021-03-07:
The UDP part of this seems to be complete with #31 (though it needs testing). Next step is to get the same for Serial.

JWhitleyWork, 2021-08-31:
This is complete.

closed 2021-08-31 by JWhitleyWork