ros-drivers/transport_drivers #14: serial_driver to be removed from Dashing due to removed dependency.

nuclearsandwich, 2020-07-14:
The removal of Autoware.Auto (which included autoware_auto_cmake) in ros/rosdistro#23937 has caused a regression of the serial_driver package in Dashing due to the dependency on autoware_auto_cmake.

There is currently a Dashing sync scheduled for this week. The current plan is to proceed with the sync and allow the package to be removed but if it's possible to work around the dependency it's possible to reintroduce the package in a future release / sync.

JWhitleyWork, 2020-07-15:
@nuclearsandwich The `autoware_auto_cmake` package shouldn't be required. I'll remove it from this build and release a new version today.

JWhitleyWork, 2020-07-15:
See #15

JWhitleyWork, 2020-07-17:
@nuclearsandwich https://github.com/ros/rosdistro/pull/25845 releases the new version into Dashing which has the Autoware.AUTO dependencies removed.

closed 2020-07-17 by JWhitleyWork