ros-drivers/transport_drivers #17: Consumers of udp_driver Fail to Link

JWhitleyWork, 2020-07-23:
Downstream packages which use `udp_driver` fail to link after 7660c2a because the dependency on the `system` module of `boost` is no longer exported.

JWhitleyWork, 2020-07-25:
Not yet fixed.

esteve, 2020-07-27:
@JWhitleyWork do you have any logs about this? I created a simple package (https://github.com/esteve/transport_drivers_consumer) that just prints the libraries that have been exported by `udp_driver` and the list seems rather sparse (only `/usr/lib/x86_64-linux-gnu/libboost_system.so`), whereas `serial_driver` exports many more libraries.

JWhitleyWork, 2020-07-27:
@esteve I've found that `serial_driver` has the issue mentioned here too. Building `xsens_node` from Autoware.AUTO with the `master` branch of this repository as a `subtree` results in:

```
Starting >>> xsens_node
--- stderr: xsens_node                                       
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::task_io_service::shutdown_service()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail15task_io_service16shutdown_serviceEv[_ZN5boost4asio6detail15task_io_service16shutdown_serviceEv]+0x7b): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
xsens_gps_node_main.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x40): undefined reference to `boost::system::generic_category()'
xsens_gps_node_main.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xb9): undefined reference to `boost::system::generic_category()'
xsens_gps_node_main.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x111): undefined reference to `boost::system::generic_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::epoll_reactor::~epoll_reactor()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactorD2Ev[_ZN5boost4asio6detail13epoll_reactorD5Ev]+0xa4): undefined reference to `boost::system::system_category()'
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactorD2Ev[_ZN5boost4asio6detail13epoll_reactorD5Ev]+0x164): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::task_io_service::~task_io_service()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail15task_io_serviceD0Ev[_ZN5boost4asio6detail15task_io_serviceD5Ev]+0x53): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::task_io_service::~task_io_service()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail15task_io_serviceD2Ev[_ZN5boost4asio6detail15task_io_serviceD5Ev]+0x53): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::epoll_reactor::shutdown_service()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactor16shutdown_serviceEv[_ZN5boost4asio6detail13epoll_reactor16shutdown_serviceEv]+0x20f): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o:xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactor16shutdown_serviceEv[_ZN5boost4asio6detail13epoll_reactor16shutdown_serviceEv]+0x27d): more undefined references to `boost::system::system_category()' follow
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
xsens_gps_node_main.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x3f): undefined reference to `boost::system::generic_category()'
xsens_gps_node_main.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xd1): undefined reference to `boost::system::generic_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::epoll_reactor::perform_io_cleanup_on_block_exit::~perform_io_cleanup_on_block_exit()':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactor32perform_io_cleanup_on_block_exitD2Ev[_ZN5boost4asio6detail13epoll_reactor32perform_io_cleanup_on_block_exitD5Ev]+0xe3): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::epoll_reactor::descriptor_state::do_complete(boost::asio::detail::task_io_service*, boost::asio::detail::task_io_service_operation*, boost::system::error_code const&, unsigned long)':
xsens_gps_node_main.cpp:(.text._ZN5boost4asio6detail13epoll_reactor16descriptor_state11do_completeEPNS1_15task_io_serviceEPNS1_25task_io_service_operationERKNS_6system10error_codeEm[_ZN5boost4asio6detail13epoll_reactor16descriptor_state11do_completeEPNS1_15task_io_serviceEPNS1_25task_io_service_operationERKNS_6system10error_codeEm]+0x254): undefined reference to `boost::system::system_category()'
CMakeFiles/xsens_gps_node_exe.dir/src/xsens_gps_node_main.cpp.o: In function `boost::asio::detail::descriptor_ops::close(int, unsigned char&, boost::system::error_code&)':
[... cut ...]
xsens_imu_node_main.cpp:(.text.startup+0x7): undefined reference to `boost::system::generic_category()'
xsens_imu_node_main.cpp:(.text.startup+0xc): undefined reference to `boost::system::system_category()'
xsens_imu_node_main.cpp:(.text.startup+0x11): undefined reference to `boost::system::system_category()'
collect2: error: ld returned 1 exit status
make[2]: *** [xsens_imu_node_exe] Error 1
make[1]: *** [CMakeFiles/xsens_imu_node_exe.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< xsens_node [20.1s, exited with code 2]
```

JWhitleyWork, 2020-07-27:
It looks like it's not a problem with this package but with how something was applied to the `xsens_node` and `velodyne_node` in Autoware.AUTO. Closing this issue and tracking in https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/557.

PR #18 (2020-07-23): Fixing boost dependency.
Closes #17.
closed 2020-07-27 by JWhitleyWork