ros-drivers/transport_drivers #26: switch to more standard default branch setup

flynneva, 2021-01-27:
right now the default branch for this repo is set to `ros2`. as far as I know this repo is only meant for ROS2....so to me there might be a better name for the default branch.

I really like the standard `develop` and `main` or `master` setup, where `develop` is where active development happens & `main` or `master` has the most recent tested & released code.

what do you think?

JWhitleyWork, 2021-01-27:
I'm fine with `main` (there's been a big push to move away from `master` lately both in ROS and GIthub in general due to the negative connotations) and `develop`.

JWhitleyWork, 2021-01-27:
Done. And re-targetd your MR to `develop`.

JWhitleyWork, 2021-01-27:
https://github.com/ros/rosdistro/pull/28154

closed 2021-01-27 by JWhitleyWork