ros-drivers/transport_drivers #29: Make downstream drivers component-able

esteve, 2021-02-08:
It should be possible to create downstream drivers as components, i.e. registered in the component registry, no main function, etc.

This ticket is part of https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/277

flynneva, 2021-03-09:
i think this can be closed, right @esteve @JWhitleyWork

JWhitleyWork, 2021-03-09:
@flynneva Since UdpSocket is not dependent upon ROS directly, yes, it is possible for `udp_driver`. However, the way that `serial_driver` is currently written, it requires inheritence from a `SerialDriverNode` class which is ROS-dependent and has a `run` function which isn't started on construction. This makes it difficult to work with in a Component. If we update the structure of `serial_driver` to match that of `udp_driver`, then yes, this can be closed.

flynneva, 2021-04-08:
@JWhitleyWork can this one be closed now?

JWhitleyWork, 2021-04-08:
Yes. Closing.

closed 2021-04-08 by JWhitleyWork