ros-drivers/transport_drivers #48: replace MutBuffer with std::vector<uint8_t>

flynneva, 2021-03-25:
referenced on #40 comments since #40 begins to migrate towards `std::vector<uint8_t>` to fix some race conditions. lets go ahead and replace the rest of the `MutBuffers` with `std::vector<uint8_t>` to keep it all consistent.

reza-ebrahimi, 2021-03-25:
I read #40 comments, but I'm really didn't get the exact reason of changing MutBuffer to a vector of bytes?

flynneva, 2021-03-25:
@reza-ebrahimi mutable buffers do not own the underlying data that they refer to. This caused us to see some threading issues where the buffers were pointing to locations in the memory that no longer had the data we wanted. I tried showing it in the print outs on #40

flynneva, 2021-03-25:
Also there is some added benefits since we can use a vector of bytes right in 'udp_socket' as the buffer to fill - which makes it super easy to then parse out into different messages in our 'to_msg' and 'from_msg' functions

JWhitleyWork, 2021-03-25:
Done for `serial_driver` in #47. Will create a separate MR to update `udp_driver` to match once #47 is merged.

reza-ebrahimi, 2021-03-26:
Why we need the buffer to own the underlying data? Also you mentioned about threading issues, IMO if there are such problems, there should be a better way to fix them.

Mutable buffers are accepting both primitive and POD types with zero copy of the data. Your changes do copy of the data to vector of bytes which is very expensive operation and totally unnecessary.

flynneva, 2021-03-26:
@reza-ebrahimi if you know of a better way to fix it please make a PR!

flynneva, 2021-04-08:
closed with #50

closed 2021-04-08 by flynneva