ros-drivers/transport_drivers #52: first release?

flynneva, 2021-04-01:
would it be worth while to try and get at least the udp_driver package released to foxy and rolling this week since next week (I think) is when rolling will be frozen for the first galactic release?

flynneva, 2021-04-29:
@JWhitleyWork any word on this? what is the blocker here for getting a beta released?

JWhitleyWork, 2021-05-01:
@flynneva Someone I was working with recently found a critical bug in the way that `to_msg` and `from_msg` were implemented for the serial driver. When fixing this bug, I found a much more concise way to implement those functions. Unfortunately, this now causes timeouts in the UdpDriver tests. Would you mind taking a look at the `unify-msg-converters` branch?

flynneva, 2021-05-01:
following up on this over topic at #55

esteve, 2021-07-13:
@JWhitleyWork @flynneva is there anything I can help with to get `transport-drivers` released?

flynneva, 2021-07-13:
@esteve my understanding is there is still the open item about the changes on `unify-msg-converters` breaks the udp logic. help with debugging/fixing that would probably be the most beneficial. see #55 for more information when I tried to fix it.

JWhitleyWork, 2021-08-31:
This is now done (except for rolling due to an upstream bug). See #61.

closed 2021-08-31 by JWhitleyWork