ros-drivers/transport_drivers #53: building shared lib on windows

flynneva, 2021-04-13:
I ran into this issue while trying to write a similar package that uses a shared lib like `io_context`.  In order to get the lib to build on Windows I needed to add the following to the beginning of the shared lib `CMakelists.txt` file.

```
## needed to build hfl_lib on windows
if(MSVC)
set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS TRUE)
set(BUILD_SHARED_LIBS TRUE)
endif()
```

not sure if anyones tried to compile this on windows yet....but we might need to add this to the `io_context/CMakelists.txt` file to do so.

JWhitleyWork, 2021-04-14:
@flynneva Rather than targeting the specific compiler, can you check for `if(WIN32)`? See https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/How-To-Write-Platform-Checks for details. If this works for you, it shouldn't interrupt anyone else's workflow so I'm fine with accepting a PR for the change.

flynneva, 2021-04-23:
@JWhitleyWork wouldnt mind your input here. stuck on what I think is a problem with linking the `io_context` library when building `serial_driver` and `udp_driver`. the CI still doesnt work so ignore that, but if you could have a quick look at the changes ive made so far and share any advice it would be greatly appreciated.

https://github.com/flynneva/transport_drivers/pull/5

closed 2021-12-14 by flynneva