ros-drivers/transport_drivers #55: udp_driver tests failing on

flynneva, 2021-05-01:
so I went ahead and added some [debug prints for the udp_driver tests](https://github.com/flynneva/transport_drivers/tree/debug-unify-msg-converters) that were failing. you can see the results below.

digging into it now but I wanted to ask @JWhitleyWork if it mattered that the `udp_msgs` we are using here are not "plain" messages with **only** a `data` member. it has [some other members]...but looking at the `converters` function it looks like it **only** uses the `data` member...which should make this ok then.

```
[==========] Running 2 tests from 2 test cases.
4: [----------] Global test environment set-up.
4: [----------] 1 test from UdpSenderNodeTest
4: [ RUN      ] UdpSenderNodeTest.RosMessageToRawUdpMessageSharedContext
4: [INFO] [1619905813.391303792] [IoContext::IoContext]: Thread(s) Created: 8
4: [INFO] [1619905813.406117574] [udp_sender_node]: ip: 127.0.0.1
4: [INFO] [1619905813.406243372] [udp_sender_node]: port: 8000
4: [INFO] [1619905813.515385792] [udp_sender_node]: to be sent msg size: 2
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.515630193] [udp_sender_node]: to be sent msg size: 4
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.515711127] [udp_sender_node]: to be sent msg size: 6
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.515817287] [udp_sender_node]: to be sent msg size: 8
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.515889290] [udp_sender_node]: to be sent msg size: 10
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.515980311] [udp_sender_node]: to be sent msg size: 12
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.516043226] [udp_sender_node]: to be sent msg size: 14
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.516097337] [udp_sender_node]: to be sent msg size: 16
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.516204706] [udp_sender_node]: to be sent msg size: 18
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: [INFO] [1619905813.516258949] [udp_sender_node]: to be sent msg size: 20
4: [ io_context::converters::from_basic_msg ] msgtype.data size: 24
4: /home/flynn/code/ros/rolling/src/transport_drivers/udp_driver/test/test_udp_driver_nodes.cpp:117: Failure
4: Expected equality of these values:
4:   rclcpp::spin_until_future_complete( node, future_1, std::chrono::milliseconds(4000))
4:     Which is: TIMEOUT (2)
4:   rclcpp::FutureReturnCode::SUCCESS
4:     Which is: SUCCESS (0)
4: [ERROR] [1619905817.523445123] [UdpSocket::asyncReceiveHandler]: Operation aborted.
4: [  FAILED  ] UdpSenderNodeTest.RosMessageToRawUdpMessageSharedContext (4140 ms)
4: [----------] 1 test from UdpSenderNodeTest (4140 ms total)
```

JWhitleyWork, 2021-05-01:
Correct. Should be fine.

flynneva, 2021-05-07:
the only thing i can think of is that we have to initialize the template type vector member to some size before trying to pass in a message type with a `data` member of some size @JWhitleyWork.  I am doing my best to read up on templates and typenames - but I have to admit i dont know much about them so this is me just shooting in the dark.

JWhitleyWork, 2021-05-08:
Thanks for looking into it. I'll give it a shot as well.

FYI, probably one of my favorite books on C++: https://www.amazon.com/Effective-Modern-Specific-Ways-Improve/dp/1491903996

flynneva, 2021-12-14:
closing since I think this was fixed since all the tests are passing now

closed 2021-12-14 by flynneva