ros-drivers/transport_drivers #60: How to open the serial port to read serial data through the serial package

hxj0316, 2021-08-25:
Hello, I installed the serial driver through the command sudo apt-get install ros-foxy-serial-driver in the environment of ubuntu20.04+foxy, but I don’t know how I can open the serial port to read through the serial package Serial data. can you help me? Thank you~

flynneva, 2021-08-25:
the latest binaries released for both `serial_driver` and `udp_driver` do not include a launch file or executable.

I would recommend cloning the repo as-is today to get the new updates (which include executables) that you can run

JWhitleyWork, 2021-08-31:
The new version has been released but isn't yet available on the repos. Please clone the current `main` branch for Galactic or Rolling or `foxy` branch for Foxy.

closed 2021-08-31 by JWhitleyWork