ros-drivers/transport_drivers #62: New release breaks builds in Autoware.Auto

niosus, 2021-09-03:
Seems like the new release is causing the pipelines to fail. See example: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/jobs/1561321563

I can already see that the `udp_driver_node.hpp` has been removed. Do you guys think it warrants to have a list of breaking changes for the release? Ideally, also a transition guide.

JWhitleyWork, 2021-09-03:
There were only 2 organizations (that we know of) which were actively using them and this release goes from version 0.0.6 to 1.0.0 - breaking changes were intended. The paradigm of how you use the classes in these has completely changed so a list of "changes" isn't really feasible. Previously you *had* to inherit from one of the node-based classes that were included in these packages but you can now use them as stand-alone C++ classes, as shown in the example nodes.

niosus, 2021-09-06:
@JWhitleyWork could you point me to the examples you are referring to here? Are you talking about the tests?

JWhitleyWork, 2021-09-07:
Specifically https://github.com/ros-drivers/transport_drivers/blob/main/serial_driver/src/serial_bridge_node.cpp, https://github.com/ros-drivers/transport_drivers/blob/main/udp_driver/src/udp_receiver_node.cpp, and https://github.com/ros-drivers/transport_drivers/blob/main/udp_driver/src/udp_sender_node.cpp.

JWhitleyWork, 2021-11-08:
This was resolved in https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/merge_requests/1170.

closed 2021-11-08 by JWhitleyWork