ros-drivers/transport_drivers #63: No sending/receiving data on serial_driver_bridge

afonsofonbraga, 2021-09-23:
Hi, 
 I wasn't able to neither send nor receive anything with launching the  serial_driver_bridge_node.launch.py
 I'm trying to execute the launch file using the configurations of my device:
``` 
/**:
  ros__parameters:
    device_name: /dev/ttyUSB0
    baud_rate: 115200
    flow_control: none
    parity: none
    stop_bits: "1"
```
I'm using ros2 galactic in docker. I'm sending the following  message to my device :
```
echo -ne '\x00\x00\x45\x44\x76\x00\x00\x36\xFF\x4F' > /dev/ttyUSB0
```
 and I'm receiving back a message (using hexdump):
 ```
 0000000    0000    0000    0076    3600    4eff    0000    0000    0000
0000010    0000    0000    0000    0000    0000    0000    0000    0000
*
0000060    0000    0000    0000    0000    0000    0000    0000    b578
 ```
 Is there any other special config to make it work?
Thanks

autowareisgreat, 2021-10-10:
Hi,

chenjun00, 2021-12-12:
I think the problem is #72 
And I have created a PR to fix it. You can also clone my [fork repo](https://github.com/chenjun00/transport_drivers) and build from source code to use `serial_driver` for now.

afonsofonbraga, 2021-12-14:
Thanks for the reply! I've decided to create one by hand and it has worked. I'll try your fork later.

closed 2021-12-14 by afonsofonbraga