ros-drivers/transport_drivers #79: UDP Bridge Node is not working

kkomurcu, 2022-04-01:
Hello Community and @flynneva,
I have a problem while running the bridge node. I can see ip address and port but cannot see message in terminal after 
running this terminal code:
 
`ros2 run udp_driver udp_bridge_node_exe --ros-args --params-file ./src/transport_drivers/udp_driver/params/example_udp_params.yaml `

The figure about this problem is below. Can you help me about this?

[image]

Pc: Msi gs66 stealth
Os: Ubuntu 20.04
Ros Distribution: Ros2 foxy

Thanks and regards.

flynneva, 2022-04-04:
hi @kkomurcu, could you include a bit more in what is going wrong / what you are seeing?

The screenshot you included seems to be the normal output I think?

closed 2022-12-01 by JWhitleyWork

Myung-Hyun, 2024-04-26:
I have same issue in my situation. I don't have any problem, I mean upb_bridege_node don't die, but the node don't publish any data even if packets is on the ethernet network. 

How can I do to solve this??

flynneva, 2024-04-30:
@Myung-Hyun I'd recommend opening a new issue and linking to this one. Maybe try some of the following to debug:

1. Can you see the node in `rqt_graph` or `ros2 node list` CLI?
2. Does `ros2 topic echo` work for the topic you are publishing to?
3. Are you accidentally setting the `ROS_DOMAIN_ID` somehow, causing the nodes topic to be published to a different domain?

Any of those might point you to the issue...but its hard to debug with the info you provided. If you still have an issue, open a new issue and share more detailed info on exactly what the issue is and what you have tried to fix it 👍🏼