ros-drivers/transport_drivers #97: can't read from the serial port. Publsiher is not activated.

Rak-r, 2024-02-04:
I am trying to read from the serial port but after running the launh file , terminals outputs Publisher is not activated. I tried to check that port is available and about the permissions too using `sudo dmesg | grep ttyACM0`, but still the same. However, I oberved that whenever i run the `dmesg` command the terminal console the same port `/dev/ttyACM0` comes listed `5-6`rows, which means that the serial port previously when the normal test script ran is not cleaned after terminating or remianed block. I have been looking and trying to find the solution but hard luck. It would be great if any could help with this.  @sven-hoek

sven-hoek, 2024-02-04:
Errrm I'm not really a maintainer of this package but

- What node are you trying to launch with which command?
- What's the output of it?
- What's the exact output of dmesg?

Rak-r, 2024-02-05:
[image]
When I try to do `sudo dmesg | grep ttyACM*` its just list like the image above. 

Terminal output

`[INFO] [launch]: All log files can be found below /home/rakshit/.ros/log/2024-02-05-12-52-02-255307-rakshit-IdeaPad-Gaming-3-15IMH05-11675
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/humble/lib/python3.10/site-packages/launch_ros/events/lifecycle/lifecycle_node_matchers.py:30: UserWarning: 'matches_node_name' has been moved into the 'launch.events' module and will be removed from the 'lifecycle' module in the future
  warnings.warn(
[INFO] [serial_bridge-1]: process started with pid [11686]
[serial_bridge-1] [INFO] [1707137522.391581589] [IoContext::IoContext]: Thread(s) Created: 2
[serial_bridge-1] [WARN] [1707137522.629742256] [LifecyclePublisher]: Trying to publish message on the topic '/serial_read', but the publisher is not activated

`

I wonder sometimes i see the data published on the topic and when i unplug the device it should stop which is ok. After agian running the `serial_bridge_node` it starts with above terminal log but `/serial_read` topic still outputs data and sometimes there is nothing. I have no clue what is happening. I tried with c++ script and python script using libserial and pyserial, it reads sometimes but when i unplug the senor and plug it back in it stops sending the data.@sven-hoek

Rak-r, 2024-07-16:
Sorry for the delayed response but I figured out the problem. It was not my custom ROS2 node talking via serial to sensor. It was the Teensy board compatibility issues with Ubuntu. I needed to add the udev rules correctly to my system and it worked flawless. Closing this issue.

closed 2024-07-16 by Rak-r