ros-drivers/transport_drivers #101: [Jazzy] Could not find a package configuration file provided by "asio_cmake_module"

Timple, 2024-07-23:
In `io_context` the `asio_cmake_module` is listed as `buildtool_depend`: https://github.com/ros-drivers/transport_drivers/blob/9fff59f66e4e0f9296501b3f671adc6543509996/io_context/package.xml#L14

So probably the buildfarm doesn't add this as dependency so it's not installed on a users system which only has `io_context` as dependency.

However it should be `find_package`d as seen here: https://github.com/ros-drivers/transport_drivers/blob/9fff59f66e4e0f9296501b3f671adc6543509996/io_context/io_context-extras.cmake#L17

I don't think it's required in `io_context-extras.cmake`, what do you think?

Hzx158, 2024-08-12:
I also encountered the same problem. Have you solved it

Timple, 2024-08-12:
I think I just manually installed it for now:
```
apt install ros-jazzy-asio-cmake-module
```

jjannik, 2024-09-27:
This also happened to me today on a ROS2 Humble.

The issue was **solved** when installing
`apt install ros-humble-asio-cmake-module`

**Here is what I tried to do**
The cmake file `io_context-extras.cmake` which was not there in an older artifact: 
`ros-humble-io-context                              1.2.0-2jammy.20230919.223322`
is available in the current artifact
`ros-humble-io-context                              1.2.0-2jammy.20240729.001626`

Which has a dependency to asio_cmake_module
`find_package(asio_cmake_module REQUIRED)`

which makes `ros-humble-asio-cmake-module` a dependent package of 
` ros-humble-io-context `

However this is not correctly reflected inside the Debian Package of `ros-humble-io-context`

Timple, 2026-01-14:
Bi-yearly ping

iskarian, 2026-10-26:
There are currently two pending PRs to resolve this issue: https://github.com/ros-drivers/transport_drivers/pull/116 https://github.com/ros-drivers/transport_drivers/pull/115